mooltipass/minible #24: Bluetooth freezing after some time.

Timandrias, 2019-11-14:
### Expected behavior

I expect bluetooth to work (not really :p)

### Actual behavior

When I connect bluetooth to the computer, sometimes, after some time it disconnects. When I try to disable bluetooth in the device, it freezes.

### Step by step guide to reproduce the problem

Connect to bluetooth using the "add bluetooth or other device" assistant on windows.
It appears as "Connected" in the bluetooth device list. Moolticute detects it.
After some time, it appears just as "Paired". Moolticute ceases to detect it.
Trying to add it again does not work
If, in the bluetooth menu of the device, I click on the "Disable Bluetooth" button, Mooltipass freezes.

### Moolticute Version - If Involved

v0.42.5-testing

### Operating System
- Windows 10
  
Other operating systems are not supported.

### Mooltipass Extension

limpkin, 2019-11-14:
What I believe is happening: the AUX MCU crashes whenever a Bluetooth disconnection happens. Whenever the main MCU needs to talk to the AUX (disabling Bluetooth, going to sleep), an infinite loop will occurs as the AUX won't be able to answer.  
This leads to high current consumption (as the platform can't go to sleep) and device freeze whenever an action requiring the aux MCU is required.

limpkin, 2019-11-24:
I believe this has been fixed!

closed 2019-11-24 by limpkin