mooltipass/minible #33: Add an "All categories" item

elendil-software, 2019-11-15:
### Missing feature

Having the ability to put credentials in categories and show credentials of a specific category is nice. It would be even better to have a category "All categories" that would display all credential without any filter

### Justification
Sometime it is quite annoying having to change categories every time.
 
### Workarounds

Need to select the correct category

limpkin, 2019-11-15:
hmmm that's an interesting point... could you elaborate on "Sometime it is quite annoying having to change categories every time." ?

elendil-software, 2019-11-16:
Example : 

I have some credential I have to enter manually. For example credential for a network share from my NAS. I put these kind of credentials in a specific category to make them easily reachable manually. So I select the category when I need to connect the share and then select the Default Categrory where I have all other credential.
But this credential is also the one I use to connect to the NAS web interface. So to connect to this interface in need to select the specific category again before selecting the Default Category, again.

Having an "All category" which would "contains" the credential of all categories would avoid to switch 
between categories when I don't need to (credential entered automatically by plugin).

limpkin, 2019-11-17:
Should I simply change the "default category" to "all categories" in that case?
Or is there a real need for a default category / all categories / category 1.... ?

elendil-software, 2019-11-17:
I think that would be even simpler.

limpkin, 2019-11-17:
in the "no category".

In details: each credential in our database can be tagged with 0, 1, 2, 3, 4.
When no category is specified, they are tagged with 0. 
tags 1 2 3 4 have their dedicated category names.
When you'd select all category, the device won't apply any filter.

I can easily implement it if that's alright with you

elendil-software, 2019-11-17:
Ok 👍

limpkin, 2019-11-17:
implemented in ogic change: now using an "all categories" filter instead of "default category"
use the bundle.img (flash main mcu) attached to this comment if you'd like to test it in advance of the release :)
[bundle.zip](https://github.com/mooltipass/minible/files/3856144/bundle.zip)

closed 2019-11-17 by limpkin

elendil-software, 2019-11-18:
Perfect ! 👍