mooltipass/minible #36: Minor usability issue

dehne, 2019-11-18:
### Expected behavior
The icon for the currently selected menu item in the Bluetooth menu should be more or less centered over its labeling text.

### Actual behavior
Unlike the other menus, the icon for the currently selected menu item is shifted to the right in the Bluetooth menu, making it less than clear which menu item is the current one.

### Step by step guide to reproduce the problem
1. Unlock the minible by inserting the card and entering the PIN.
1. Navigate to the "Enable Advanced Menu" menu item in the Operations menu.
1. Observe that the icon is centered over the text under it.
1. Navigate to "Pair New Device" item in the Bluetooth menu.
1. Observe that the currently selected icon (the B+ item) is well to the right of the text under it.

### Moolticute Version - If Involved
N/A

### Operating System
- Windows 7
  
Other operating systems are not supported.

### Mooltipass Extension
N/A

limpkin, 2019-11-18:
Yep.... which is why i try to keep the number of items to 5 or 7 on the carousel :D

limpkin, 2020-02-19:
upcoming firmware will have 5 items :D

closed 2020-02-19 by limpkin