mooltipass/minible #38: Rotary encoder skips steps when rotating fast

WebmasterTD, 2019-11-19:
### Expected behavior

The encoder should follow the rotation precisely even at higher rotation speeds.

### Actual behavior

Skips steps and even misses the direction when the encoder rotates too fast.

### Step by step guide to reproduce the problem

Just rotate the encoder fast enough on any menu screen or pin input.

_(Might be problem with the input encoder software debouncing algorithm)_

limpkin, 2019-11-19:
What's the use case that justifies this need though?

WebmasterTD, 2019-11-19:
I feel like I could input my pin faster in the Mooltipass Mini. Scrolling through the menu items is not that big of a problem, but once the credentials list grows bigger it would certainly help to make it quicker.  I believe this could make the BLE feel more responsive. 

Also maybe speeding up the animations in some places could help too.

limpkin, 2019-11-20:
Let's wait and see if others share your opinion :)
You can click to skip any animation!

limpkin, 2019-11-23:
PIN animation needs to be sped up

limpkin, 2019-11-23:
with this commit, a (much) faster pin animation is enabled. Please let me know if there's another animation that you feel is too slow :)  
If you want to try the updated firmware before the official firmware update, use the attached file.
[bundle.zip](https://github.com/mooltipass/minible/files/3882777/bundle.zip)

pinging @ghalfacree as well

WebmasterTD, 2019-11-24:
It definitely feels better now.
One thing - when I click at the "Going to sleep..." animation it goes to sleep instead of waking up.

limpkin, 2019-11-24:
You asked, we delivered :)
You can try this bundle file: 
[bundle.zip](https://github.com/mooltipass/minible/files/3883823/bundle.zip)  

I'll close this issue for now as it's a bit out of scope, but please feel free to make a new one for improvements you would have in mind.... this is **exactly** what we'd like to hear from our beta testers!

closed 2019-11-24 by limpkin

ghalfacree, 2019-12-16:
Apologies for dredging this up from the depths, but I've only just had a chance to update the firmware - the inability to import credentials from my MPM means that I'm not using the BLE as a daily driver, and can only test it every now and again.

I've updated to 0.14/0.10, and the PIN entry is still a lot slower than on the MPM. Ideally, I'd like to see the option to disable the "smooth scrolling" animation altogether - so that one click of the wheel immediately switches to the next character, no delay. Same for other scrolling operations - not so much the main menu, because there are only a handful of choices there, but definitely when scrolling through the credentials list there should be an option to disable smooth scrolling.