mooltipass/minible #40: Outputs plus-minus [ ± ] instead of pipe [ | ]

the-other-foot, 2019-11-20:
I haven't tested this with usernames, but passwords that contain a pipe character `|` show up correctly in credential management, display correctly on the device, but output as the plus-minus character `±` instead. I haven't yet tested what it outputs if the password contains the `±`  though.

The Aux MCU version is 0.8, the Main MCU version is 0.7

moolticute and the device both report language as English US, and Bluetooth+USB layouts as USA and I'm only using US-based input devices with macos.

### Expected behavior

Outputs the password including pipe characters `|`

### Actual behavior

Outputs the password substituting pipe characters `|` with plus-minus characters `±`

### Step by step guide to reproduce the problem

- store a credential with a pipe character included in the password
- find the credential and output it [in this case over usb; bluetooth untested]

### Moolticute Version - If Involved
0.42.8-testing

### Operating System
macos 10.14.6

limpkin, 2019-11-20:
This is caused by the fact that macOS layouts aren't added to the device yet.

limpkin, 2019-11-23:
going to close this one as macos layouts have been implemented.

closed 2019-11-23 by limpkin