mooltipass/minible #43: sending username/password with BT under OS X (set to 20ms delay)

mpble, 2019-11-23:
Use one of the two templates below and delete the other. All issues not following this template will be directly closed.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior
Username and password will be entered from Mooltipass BLE at the position of the cursor in the active edit field.
### Actual behavior
Username and password are not entered at the position of the cursor in the active edit field. 
### Step by step guide to reproduce the problem
- Connect Mooltipass BLE with MacBook Pro with BT.
- select under MacOS a position, where you want to insert the username.
- select Login on Mooltipass BLE
- select an entry type Login, confirm Ok
- select under MacOS a position, where you want to insert the password
- confirm Ok

### Moolticute Version - If Involved
0.42.9
### Operating System
MacBook Pro, Model 2017, MacOS Catalina (10.15.1)
Mention if you are using either:
- Windows 7/10
- Debian Based distribution
- Other Linux distribution
- MacOS  
  
Other operating systems are not supported.

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2019-11-23:
"Username and password are not entered at the position of the cursor in the active edit field."

are they entered at all?

limpkin, 2020-04-19:
closing issue due to outdated code and inactivity
new issue to be opened if bug still exists

closed 2020-04-19 by limpkin