mooltipass/minible #48: Mooltipass Mini BLE unresponsive after BLE usage

ThunderKing54, 2019-11-28:
### Expected behavior
I connected my Mooltipass BLE mini with Bluetooth and used it for 10-20 minutes. After that I pulled out the card and left the device on my table for 2-3 hours. The Mooltipass BLE mini should wake up when I insert my smartcard, press the scroll wheel or if I use the USB-C kabel.
### Actual behavior
The Mooltipass BLE mini wasn't responding anymore. Only after pressing the scroll wheel for a long time I saw the boot animation. After it was responsive again I saw that the battery was fully charged
### Step by step guide to reproduce the problem
I couldn't reproduce the problem yet. But these steps might caused the problem.
- charge the mooltipass to 100%
- unplug the mooltipass and connect it with a other computer(Bluetooth)
- Pair the Mooltipass with your PC(Bluetooth)
- use it for a while
- shutdown your PC
- unplug the smartcard and leave the mooltipass on the table for 2-3 hours
- try to wake it up

### Moolticute Version - If Involved
v0.42.14-testing

### Operating System
Windows 10 18362

### Mooltipass firmware version
Aux MCU 0.9
Main MCU 0.9

limpkin, 2019-11-28:
my guess is that bluetooth froze on the device... will investigate asap (next week after finding that usb bug)

limpkin, 2019-12-18:
btw is this still happening?

ThunderKing54, 2019-12-19:
I try to reproduce this problem. I close this issue if I can't find anything until Friday night.

closed 2019-12-21 by ThunderKing54