mooltipass/minible #51: Paired device does not reconnect

elendil-software, 2019-12-01:
### Expected behavior

If the device has been paired to a computer, it should automatically reconnect.

### Actual behavior

The device does nor reconnect especially when it has been disconnected for a long time. This does not to happens when the device is reconnected only a few minutes after being disconnected 

### Step by step guide to reproduce the problem

1. Pair the device
2. Disconnect the device by disabling Bluetooth on the device or computer
3. Later, enable the Bluetooth again
4. The device is not connected it only show the bluetooth is enabled. On the computer it is shown as paired, but not connected 

### Moolticute Version

- 0.42.14-testing

### Operating System

- Windows 10

the-other-foot, 2019-12-06:
I'm encountering this as well on my mac. 

If I remove the mooltipass OS-side from my bluetooth devices, I can re-pair it as a keyboard [though it doesn't output any text (#53)], but when I disable bluetooth on the mooltipass and re-enable, it appears to try to re-pair but fail continuously.

### Expected Behavior
mooltipass reconnects to previously paired system after disabling/re-enabling bluetooth

### Actual Behavior

mooltipass continuously fails to pair

### Step by step guide to reproduce the problem
- remove device pairing configuration from OS side
- enable bluetooth on mooltipass
- pair mooltipass with OS
- disable bluetooth on mooltipass
- enable bluetooth on mooltipass

### Operating System
macos 10.14

### Firmware Versions
aux: 0.10
main: 0.10

limpkin, 2019-12-18:
just to confirm, what aux & main mcu version does this apply to?

limpkin, 2019-12-18:
@elendil-software when you reported this bug, what aux MCU version was it?

elendil-software, 2019-12-19:
@limpkin I will have to check. Probably the last available version available at that time. I always do the update as soon as  you send a new version.

Julien

elendil-software, 2019-12-20:
@limpkin I've done new test and I still have the same kind of problems with Aux MCU 0.10 and Main MCU 0.13

limpkin, 2020-02-29:
that is now fixed :)

closed 2020-02-29 by limpkin