mooltipass/minible #52: Incorrect output for USA (MacOS) Keyboard output

the-other-foot, 2019-12-06:
when outputting over usb, text encoding is incorrect for the USA (MacOS) layout on a mac. I've validated that the keyboard layout for usb reads USA (MacOS) from both Moolticute and the device's settings menu.

### Expected behavior
username `abcdefg1234567890` and password `1!aAqQpP4$'"` would output as
```
abcdefg1234567890	1!aAqQpP4$'"
````
### Actual behavior
username `abcdefg1234567890` and password `1!aAqQpP4$'"` outputs as
```
]^_` abc-./012345,	-8˝]=mMlL0 #
```
### Step by step guide to reproduce the problem
- Set keyboard layout to USA (MacOS)
- Add any credential
- Send credential out over USB

### Moolticute Version
0.42.20

### Operating System
macos 10.14

### Firmware Versions
Aux: 0.10
Main: 0.10

limpkin, 2019-12-06:
could you try US Extended (MacOS)?

the-other-foot, 2019-12-06:
Aha! Yep, US Extended (MacOS) works as expected, though I'm surprised at it using US Extended.

limpkin, 2019-12-06:
Would you know by any chance the difference? :)
Are you in the US? I'll setup my mac soon....

limpkin, 2019-12-12:
pinging @the-other-foot

limpkin, 2019-12-15:
@the-other-foot  ?

the-other-foot, 2019-12-16:
oop sorry for the delayed response! been a bit hectic around here recently :)

I'm afraid I don't know the difference, though I'd guess the extended keyset would support characters outside the standard US 102(ish?) keyset. And yep, I am in the US.

I also noticed that while set to US Extended (MacOS), it also outputs a space character after several punctuation marks including `.`, `,`,`-`, and backtick; e.g. a password stored as `foo.bar` will display correctly as `foo.bar`  on the device and recall correctly as `foo.bar` in moolticute but be output as `foo. bar`

limpkin, 2019-12-18:
that's quite interesting! could you file a new bug just for that @the-other-foot ?

limpkin, 2020-05-02:
fixed by 8818d3a8d73b001391d68acedb4d4486fdbdf115

closed 2020-05-02 by limpkin