mooltipass/minible #53: No text output over Bluetooth on mac

the-other-foot, 2019-12-06:
When the mooltipass is connected via bluetooth [with and without the USB cable also connected], it doesn't output any text. After sending the password, the mooltipass's screen does "pause" like normal as if it's busy sending the data, but nothing gets typed.

### Expected behavior
Credentials are outputted on screen

### Actual behavior
No output appears

### Step by step guide to reproduce the problem
- pair mooltipass via bluetooth
- attempt to send credentials

### Moolticute Version - If Involved
0.42.20

### Operating System
macos 10.14
  
### Firmware Versions
Aux: 0.10
Main: 0.10

arieljt, 2019-12-17:
I've been experiencing the same, any update on that?

limpkin, 2020-04-19:
upcoming release to fix this!

limpkin, 2020-04-19:
fixed in 18e7a424c04db4e70366a7c88e5faccfe3f43764

closed 2020-04-19 by limpkin