mooltipass/minible #55: Mooltipass Mini BLE crashes during credentials recall

ThunderKing54, 2019-12-09:
### Expected behavior
Mooltipass Mini BLE shouldn't crash after it wants that I confirm the credentials for a specific website(gitlab.northernroot.net).
### Actual behavior
The Mooltipass Mini BLE crashes after visiting a specific website(gitlab.northernroot.net).
### Step by step guide to reproduce the problem
First I connected the Mooltipass with USB. After that I changed the Keyboard Layout with Moolticute to Germany and saved the settings(no card inserted). Then I inserted my key card into the Mooltipass and confirmed the pin. After the pin was accepted I went to the Bluetooth settings menu. I checked if bluetooth was enabled or not. Bluetooth wasn't enabled. I visited my cloud and confirmed the credentials. The Mooltipass inserted the credentials and I was logged in. After that I visited my gitlab website. The Mooltipass wanted that I confirm the credentials again but it allready crashed at this time.

I tried a forced shutdown but it didn't helped.
https://www.youtube.com/watch?v=g-_SaJcDPT8

After I stopped recording I pulled out the USB cable again and plugged it back in. The Mooltipass displayed the message: No Aux MCU
Windows notified me that the connected device wasn't reconised. I pressed the scroll wheel again for 15 seconds and the message disappeared. After that the Mooltipass started again.

[image]
[image]

### Moolticute Version
v0.42.23-testing

### Mooltipass firmware
Aux MCU version: 0.10
Main MCU version: 0.11 (customized for: Card Failing! message)

### Operating System
Windows 10 Pro 18362

limpkin, 2019-12-09:
Is bluetooth enabled?

ThunderKing54, 2019-12-09:
I checked it and it was disabled. It is also visible inside the video. I tried something else a minute ago. My mooltipass even crashes when I go to "Credentials" in Moolticute and try to view the credentials for gitlab.northernroot.net

limpkin, 2019-12-09:
ok... can you confirm it's just this site and nothing else?

ThunderKing54, 2019-12-09:
This problem also occurs on accounts.google.com. I can't really test everything. It takes 1-2 hours until the Mooltipass battery is empty again. Should I try a older firmware and Moolticute version? This problem started after I flashed Main MCU version 0.11.

limpkin, 2019-12-10:
in order to allow faster iteration, could you remove the USB cable before you long press the button?  
that should allow you to force shutdown the device.
trying an older firmware could be interesting indeed... thanks!

ThunderKing54, 2019-12-10:
I can't force shutdown the device. 

press scroll wheel for 15 seconds: nothing happens
press scroll wheel for 15 seconds and press ist again after that for 1-2 seconds: nothing happens
press scroll wheel for 2 seconds: nothing happens
press scroll wheel for >15 seconds: nothing happens

It only works if the battery is empty. I flashed the firmware version 0.10 for Aux and Main MCU. Same problem occurs. Next I gonna install Moolticute Version v0.42.18-testing. If this version works I update the firmware version back to 0.11 for Main MCU. The wired behaviour started after I flashed Main MCU 0.11. The Mooltipass worked but Moolticute was crashing all the time. Raoul fixed it after he published Moolticute version 0.42.20-testing. But then my Mooltipass crashed all time.

ThunderKing54, 2019-12-10:
The Mooltipass crashes as well with Moolticute version 0.42.18-testing and Aux MCU, Main MCU version 0.10. I try to flash a older firmware later. What if I create a account for you on gitlab.northernroot.net(my gitlab) with similar credentials. Maybe you can repoduce the problem?

ThunderKing54, 2019-12-10:
Hello Mathieu I found the main reason why this happen. If you enable simple mode my Mooltipass start crashing when visiting gitlab.northernroot.net. But when I enable the advanced mode it doesn't happen. It might be a timing issue? That also explains why it doesn't happen earlier to me. I saw the conversation about filling out passwords with the Login menu. So I switched to the simple mode some days ago but didn't visited gitlab.northernroot.net or accounts.google.com.

limpkin, 2019-12-10:
hmmmm would you be able to share privately your email address that you use for these accounts? I can't manage to reproduce this issue in simple mode

limpkin, 2019-12-10:
nevermind i seem to have found a way to reproduce it! will keep you updated

closed 2019-12-10 by limpkin

limpkin, 2019-12-11:
@ThunderKing54 : I wanted to thank you for the stellar bug report. Your clear explanation and follow-up were crucial to getting this bug fixed... it's a pleasure having you as a beta tester :)

commit 3728d0d8 (2019-12-10): fix #55