mooltipass/minible #61: Windows 10 Cycles between Paired and Connected

dehne, 2019-12-19:
## Expected behavior
Once a minible has been paired and connected with Windows 10, disabling and then re-enabling Bluetooth on the device should result in the minible reconnecting.
### Actual behavior
Instead of successfully reconnecting, Windows shows the status cycling between "Connected" and "Paired" with a period of between one and two seconds.
### Step by step guide to reproduce the problem
1. Bring up the Windows 10 "Bluetooth & other devices" panel.
1. Pair and connect the minible to the Windows system in the usual way.
1. Once Windows shows the device connected, disable Bluetooth using the device's menu.
1. Wait for Windows to change the status from "Connected" "Paired".
1. Enable Bluetooth on the minible using the device's menu
1. Windows will now display the device's status cycling back and forth between "Paired" and "Connected".
### Moolticute Version - If Involved
Not involved.
### Firmware Version
AUX MCU version:  v0.10
Main MCU version: v0.14 

### Operating System
- Windows 10

limpkin, 2020-02-29:
@dehne is this still happening with the latest firmware update? :)

dehne, 2020-02-29:
Tested with AUX MCU v0.14, Main MCU v0.19. Now works as expected. Yay!

closed 2020-02-29 by dehne