mooltipass/minible #62: Feature Request: Disable pair options when Bluetooth is off

ghalfacree, 2019-12-30:
8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature
When you enter the Bluetooth menu on the MBLE, the default option is "Pair new device." Choosing this, however, does nothing - because, by default, Bluetooth is switched off. You have to scroll across to "Enable Bluetooth," *then* attempt to pair a new device.

It would be smoother to disable the pair option (and, possibly, unpair) until Bluetooth is enabled. Alternative solutions would be to make "Enable Bluetooth" the default option if it's not already enabled (so you see that it's disabled before being given the option to pair) or to have the pair icon bring up an error with the option to enable Bluetooth.

### Justification
From a first-user perspective, choosing "Pair new device" and having nothing happen feels like - though technically isn't - a bug.

### Workarounds
Only to make sure you enable Bluetooth before attempting to pair a device!

8<------------------------ END FEATURE REQUEST --------------------------------

ghalfacree, 2019-12-30:
It may also be nice to have the "Pair new device" icon bring up a message when you press the scroll wheel, like "Search for Mooltipass BLE", as it's currently not clear that just having the icon selected but not activated with a click is good enough to begin the pairing process.

limpkin, 2020-01-04:
partly implemented in 2f179f3b8bdc23455840858f661f259d93a14ae5  :)

I'll however close this issue as the pair / unpair options haven't been implemented yet :)

closed 2020-01-04 by limpkin