mooltipass/minible #65: Battery indicator shows wrong value

ThunderKing54, 2020-01-08:
### Expected behavior
If I connect my Mooltipass Mini BLE with USB it should show the same Battery value like before when I disconnect it again(5-10 seconds). 
### Actual behavior
If I reboot my Mooltipass it shows three Battery bars. The Mooltipass is not connected with Bluetooth or USB. After I plug the USB cable into my Mooltipass it starts charging. When I disconnect my Mooltipass again it only shows one Battery bar. 
### Step by step guide to reproduce the problem
1. Use your Mooltipass a whole day only with USB.
2. Disconnect the Mooltipass after 8 hours and leave it for 15 hours inside your backpack. 
3. Insert your smartcard and look at the Battery bars. It should have 4 to 3 bars. 
4. Connect your Mooltipass with USB to the PC(USB-Ports on the back side of the PC).
5. The Mooltipass start charging.
6. Disconnect the Mooltipass after 5-10 seconds. 
7. The Mooltipass now shows only 1 Battery bar for me. If I leave it for multiple minutes it shows more and more Battery bars until it is charged. It is fully charged after about 5 minutes.
### Moolticute Version - If Involved
v0.42.26-testing

### Firmware Version
AUX MCU version:  0.11
Main MCU version:  0.16

### Operating System
Windows 10 Enterprise 2016 LTSB

limpkin, 2020-01-08:
Thanks for the bug report! I'll look into it :)

ThunderKing54, 2020-01-08:
I worked with the Mooltipass mini BLE(Bluetooth enabled) for about 7 hours now. The battery drained pretty fast(one battery bar left).

Bat: ADC 3257 1265mV

limpkin, 2020-01-08:
Yep.... that's expected as we need to implement a feature that should make this a few days :)

limpkin, 2020-01-08:
Hello!

Could you confirm this has disappeared with the attached firmware?
[bundle.zip](https://github.com/mooltipass/minible/files/4037116/bundle.zip)

closed 2020-01-08 by limpkin

ThunderKing54, 2020-01-09:
I tested it a bit and I think the bug is fixed. There is still a small beauty issue. If I reboot my Mooltipass it shows 3 battery bars and after 1 second or even less it shows the "real" battery value without a USB cabel is connected. So it is refeshing the battery status now but is still not fast enough. It should show the current battery status after the boot animation finished.

limpkin, 2020-01-09:
Yep! I've noticed that as well and it should be fixed in 872557884d41162fda55597351b4aecc6b3fbba1

commit c2f08ead (2020-01-08): fix #65