mooltipass/minible #66: BLE doesn't charge from a wall charger - only PC

ghalfacree, 2020-01-08:
8<------------------------ BUG REPORT -----------------------------------------
### Expected behaviour
When the BLE is connected to a "dumb" wall charger - in my case, an IKEA desk lamp with a USB port on the bottom - it shows the battery charging animation and, you would expect, charges its battery.

### Actual behaviour
The battery does not seem to charge, though the animation does play. Verified by leaving it connected all day, and finding it was still on one bar of charge at the end.

Verified it further by connecting via a USB Type-C power meter, which showed the BLE drawing 0.06A while connected to the wall charger; switching it to being connected via a PC's USB port caused the BLE to draw 0.14A and charge normally.

### Step by step guide to reproduce the problem
1. Run down the BLE battery
2. Plug into a "dumb" wall charger rather than a PC or other host device.
3. Go make a cup of tea or something.
4. Come back to see if the BLE's battery is charged.

### Firmware Version
AUX MCU version:  0.11
Main MCU version:  0.16

### Operating System
N/A

8<------------------------ END BUG REPORT -------------------------------------

limpkin, 2020-01-11:
this should be solved with 0.17... could you confirm? :)

ghalfacree, 2020-01-13:
Had a quick look without running the battery down, but it didn't show anything interesting - I'll leave it sat with Bluetooth enabled for a bit and try again.

I *did*, however, manage to freeze the thing on the logo screen. Here's what I did:

* Plugged the BLE into my desktop
* Unplugged it from the desktop
* Plugged it into the dumb charger via the power meter (0.06A showing)
* Plugged it back into the desktop via the same power meter (0.03A showing)
* Frozen

Didn't respond to the card being inserted or removed, being plugged into anything, the button, nowt. Had to long-hold the button, then the screen went black; pressed it again, and it booted normally.

When the battery's run down a bit, I'll recreate the above while testing the charging and see if it was a one-off glitch or not; if not, I'll make a new issue for it.

ghalfacree, 2020-01-13:
This does indeed appear to be fixed in 0.17 (well, I tested 0.18, but let's not split hairs) - it's now drawing 0.14A from my lamp. Many thanks!

Couldn't reproduce the above screen-freezing glitch, so I'll just keep an eye out for that one happening again in the future.

closed 2020-01-13 by ghalfacree