mooltipass/minible #68: Knock detection for multiple website accounts

ThunderKing54, 2020-01-17:
### Missing feature
Knock detection on websites with multiple accounts
### Justification
There is no knock detection on website where I have multiple accounts. I have to use the scroll wheel to accept the credentials. The feature should include that you are able to switch between the accounts with a knock and you can accept the credentials for the selected account with fast tapping for 2-3 seconds.

### Workarounds
Use the scroll wheel to select the right account.

ThunderKing54, 2020-02-26:
I programmed a function for a coin acceptor once. The coin acceptor returned puleses depending on the coin value. I counted down a timer(300ms or less) and every time a pulse was send from the coin acceptor the timer was set back to 300ms again. The timer started after I inserted a coin(first pulse). If the timer stopped and no more pulses where send to reset the timer I know the value of one coin. 
1€ = 10 pulses
0,50€ = 5 pulses and so on.

That means if I knock once during 300ms the next account is displayed. If I knock more than one time during 300ms the knock counter rises and the Mooltipass know that the user knocked more than one time to accept the selected account. I don't know if your able to use this but maybe it will help.

ThunderKing54, 2020-04-02:
@limpkin new Idea.

Instead of detecting multiple knocks you could rotate though the accounts every 2 seconds(editable in settings) and if you knock during that time the current selected account credentials are send to moolticute.

limpkin, 2020-04-02:
that's a nice idea :)

commit 1193e49c (2021-02-23): storing last used service for a parent, fix #201, fix #68
closed 2021-02-23 by limpkin