mooltipass/minible #69: Remaining actions on FIDO2 implementation on main MCU

limpkin, 2020-01-19:
1) if it makes sense, not use a hmac based prng anymore... create our own one to pass to the ecc functions
- hopefully it should be as simple as tweaking hmac_drbg.ch :)
2) check for buffer overflows for ecc related functions:
- for example br_ec_keygen returns the size written into the private key buffer, we should check it with the size of our buffer and while() in case of errors

0x0ptr, 2020-01-19:
1. Why reinvent the wheel? What does it gain us?
2. Agree
3. What do you want to clean and when?
4. Yes
5. I think I answered this in the pull request. Let me know if there is anything other than that.

limpkin, 2020-01-19:
1. right now it's generating pseudo random bits and not **real** random bits
3. any context that may have been initialized (ecc perhaps? nto sure...)
 thanks again for the great PR :)

limpkin, 2020-01-23:
2 has been implemented
3 is currently not required given line 335 of logic_encryption
4 has been done
5 has been done

1 remaining

limpkin, 2020-09-05:
closing as 1. is a nice to have but not a must

closed 2020-09-05 by limpkin