mooltipass/minible #80: Black screen on disconnect usb -> battery

ThunderKing54, 2020-02-25:
### Expected behavior
If you disconnect your Mooltipass during configuration inside category, bluetooth, login menu the display should not turn black. It should stay inside the selected menu until you select something.

### Actual behavior
If you disconnect your Mooltipass during configuration inside category, bluetooth, login menu the display turn black but your still able to select a option without knowing it(black screen). If you do it the other way it works like it should(running with battery and connecting it to usb).

### Step by step guide to reproduce the problem
1) connect your Mooltipass with a USB-C cable
2) enter your smartcard and the pin code
3) select the "Set Active Category" menu and make sure your current selected category is "All Categories"
4) disconnect the mooltipass
5) you will see a black screen which won't wake up if you scroll. Scroll a bit and press the scroll button.
6) After that you see the main menu again
7) enter the  "Set Active Category" menu again.
8) now a different category is your current category now. 

### Firmware Version
AUX MCU version:  0.13
Main MCU version:  0.18

### Operating System
- Windows 10 Enterprise 2016 LTSB

ThunderKing54, 2020-03-26:
@limpkin this bug still exist inside the bluetooth enable menu.

### Step by step guide to reproduce the problem
1) insert the smartcard and enter your pin
2) enter the bluetooth menu and enable bluetooth
3) during the "Enabling Bluetooth" menu unplug the Mooltipass Mini BLE
4) screen goes black again. After a short time the message "Bluetooth Enabled" appear

limpkin, 2020-03-26:
I love how your find these bugs :D

commit 2b451d2d (2020-02-25): fix #80
commit 7ef240e8 (2020-03-26): fix #80
closed 2020-03-26 by limpkin