mooltipass/minible #87: _unlock_ feature bug

ThunderKing54, 2020-03-18:
### Expected behavior
When I insert my smartcard and enter my pin it should first press "Enter" then type in my password and then press "Enter" again when I use the _unlock_ feature with Enter + Pass / Win + L
### Actual behavior
When I insert my smartcard and enter my pin it ask me if I want to enter my credentials. When I accept it the Mooltipass press Enter then it types half of my password and press enter again. 
Example in text output:

"ENTER"
testeron
ipoharie55
"ENTER"

The example password is one word(testeronipoharie55)

### Step by step guide to reproduce the problem
1) setting up _unlock_ feature on your Mooltipass
2) Use a 18char length password
3) Enable _unlock_ feature with Option "Enter + Pass / WIN + L"
4) remove your smartcard
5) enter your smartcard and select your pin
6) accept the credential dialog 
7) Windows return with wrong password
### Moolticute Version - If Involved
v0.43.4-testing
### Firmware Version
AUX MCU version:  0.16
Main MCU version:  0.20

### Operating System
- Windows 10  Enterprise LTSB 2016

limpkin, 2020-03-18:
hello!

could you increasing the typing delay in the settings tab of moolticute? :)

Thanks!

ThunderKing54, 2020-03-18:
I increased the typing delay to 50ms. Still the same

limpkin, 2020-03-22:
try again with the latest release i sent you in private? :D

ThunderKing54, 2020-03-22:
Nothing changed. The password is still splitted :(

ThunderKing54, 2020-03-22:
But wait wait.... 
There is a letter missing when the Mooltipass writes my password. 
I gonna write you a email

limpkin, 2020-03-25:
that was a fun one to debug! 
thanks all for testing this :)
[bundle.zip](https://github.com/mooltipass/minible/files/4383967/bundle.zip)

closed 2020-03-25 by limpkin

ThunderKing54, 2020-03-26:
It works now :)
Thanks