mooltipass/minible #88: Mooltipass Mini spam key on smartcard removal with _unlock_ feature

ThunderKing54, 2020-03-18:
Hard to reproduce again :(

### Expected behavior
If I enable the _unlock_ feature with the "Win+ L" feature it should lock my pc and do nothing after that
### Actual behavior
I unplugged the smartcard and got the message "Card Failing!" again. After that the Mooltipass spammed keyboard output to the login window of windows

@limpkin I gonna send you a video again. It contains sensitive information(first name, last name and phone number) so I can't attach it here.

### Step by step guide to reproduce the problem
1) insert smartcard and enter your pin
2) pull out the smartcard sideways so the "card inserted button" might be still triggered. 
3) The Mooltipass try to read the smartcard but shows the message "Card Failing!" and Locks the computer.
4) Insert your smartcard again
5) The Mooltipass still spamming some keys(don't know which key). Maybe the L key?

### Firmware Version
AUX MCU version:  0.16
Main MCU version:  0.20

### Operating System
Mention if you are using either:
- Windows 10 Enterprise LTSB 2016

limpkin, 2020-03-18:
Thanks for the bug report!
Could you try updating your unit with the attached firmware and let me know if that fixed it?
thanks!
[bundle.zip](https://github.com/mooltipass/minible/files/4351430/bundle.zip)

ThunderKing54, 2020-03-19:
Hello Mathieu,

 

no problem! But I had some trouble to flash the image. If you see the screen “Invalid Card Inserted” you have to press the scroll wheel so you’re allowed to upload the firmware. If you don’t do that you can’t upload the image.

ThunderKing54, 2020-03-19:
Can’t reproduce it. Seems fine for me :) But I still test it a bit in production.

limpkin, 2020-03-22:
damn github... I can't receive notifications and only saw your message now :/.

I'll close it, feel free to reopen just in case!

ThunderKing54, 2020-03-22:
With the version you gave me it is not fixed. Wasn't able to repoduce it in the new version. I was able to see the spammed key

[image]

limpkin, 2020-03-25:
not sure to follow you... the last version still has the issue?
the letter being spammed... is it the last letter of your password?

ThunderKing54, 2020-03-25:
This letter is not included in my password

ThunderKing54, 2020-03-25:
I test it but I'm not able to reproduce it.

limpkin, 2020-03-25:
and the latest release also has this bug (firmware from saturday)?

ThunderKing54, 2020-03-25:
I can't tell you right know. I still try to find out if this happen again.

limpkin, 2020-03-25:
that was a fun one to debug! 
thanks all for testing this :)
[bundle.zip](https://github.com/mooltipass/minible/files/4383968/bundle.zip)

commit ca457ea2 (2020-03-25): fix #88 and #87
closed 2020-03-25 by limpkin