mooltipass/minible #89: Screen Text rotate by 180° when unplugging Mooltipass BLE mini inside password menu

ThunderKing54, 2020-03-19:
That is a good one as well. You're only able to see it for <1 second. 🙃
### Expected behavior
If you disconnect your Mooltipass BLE mini inside the password confirmation message, it should not rotate the displayed text by 180° and rotate it back after <1 second.
### Actual behavior
If you disconnect your Mooltipass BLE mini during the password confirmation mesage the screen shoes the normal text but rotate by 180° for less than 1 second
### Step by step guide to reproduce the problem
1) connect your mooltipass with USB
2) insert your smartcard and enter pin
3) go to the Login menu and select a website
4) during the "Type Login?" menu unplug the mooltipass 
5) you gonna see the same menu but rotated by 180° for less than 1 second

### Firmware Version
AUX MCU version:  0.16
Main MCU version:  0.20

### Operating System
- Windows 10 Enterprise LTSB 2016

limpkin, 2020-03-19:
holly molly that one was tricky to find... 
it's always a pleasure debugging the bugs you find.. remind me to thank you again once the mini is released!

closed 2020-03-19 by limpkin

ThunderKing54, 2020-03-20:
Thank you :) The trasure hunt starts again soon :)

commit 955baec1 (2020-03-19): fix #89