mooltipass/minible #92: Bluetooth enable animation freeze for some seconds

ThunderKing54, 2020-03-25:
It does not impact the use but it looks strange. Same happen when you flash a new bundle at the "Firmware update" notification. I looked into your source code and noticed that you circle trough the bitmap images with a loop to play animations. I don't know if it is possible to fix that issue.
### Expected behavior
If you enable bluetooth the information animation shouldn't freeze for some seconds.
### Actual behavior
If you enable bluetooth the information animation freeze for some seconds.
### Step by step guide to reproduce the problem
1) connect MBLE mini with USB
2) insert smartcard and enter your pin
3) go to the bluetooth menu and enable bluetooth
4) the animation stops until bluetooth is enabled.

### Firmware Version
AUX MCU version:  0.17
Main MCU version:  0.21

### Operating System
Mention if you are using either:
- Windows 10 Pro

limpkin, 2020-03-26:
fixed it for the bluetooth issue!
I'm afraid that I can't change the firmware update one, as during the latter i'm overwriting the graphics :)

closed 2020-03-26 by limpkin

ThunderKing54, 2020-03-27:
But a new bug appeared after that. If you unplug the mooltipass during the "Enable Bluetooth" animation it freeze the mooltipass.

limpkin, 2020-03-27:
Oh nice catch!
I've just pushed a patch for this, which has the unfortunate side effect of blanking the screen if you were to disconnect the USB cable while the bluetooth is enabling.
I'm afraid I can't think of an easy way to patch this side effect for the moment :/

commit 63a1e396 (2020-03-26): fix #92