mooltipass/minible #93: Use left handed mode rotation bug usb -> battery

ThunderKing54, 2020-03-26:
### Expected behavior
If your Mooltipass BLE mini display the message "Use Left Handed Mode?" it should stay in the rotate screen if you unplug your mooltipass

### Actual behavior
If you unplug your Mooltipass during the menu "Use Left Handed Mode?" the message rotates back to the Right Handed Mode.
### Step by step guide to reproduce the problem
1) connect your mooltipass with usb
2) enter your smartcard and pin
3) turn around the mooltipass until it diplays the left handed mode menu
4) unplug the mooltipass
5) the "Use Left Handed Mode?" menu is now displayed for right handed people ;)

### Firmware Version
AUX MCU version:  0.17 
Main MCU version:  0.21

### Operating System
- Windows 10 Pro

limpkin, 2020-03-26:
This is actually a feature (aha!), there's a left handed settings when connected to usb and another when battery powered :)

ThunderKing54, 2020-03-26:
Yes it is buuuuut! If you rotate your mooltipass it ask if you want to switch to the Left handed mode and displays the message so it is readable in left handed mode. If you unplug the mooltipass it still shows the menu but the message is now rotated back and you can't read it ;D Maybe i should do a video about it.

limpkin, 2020-03-27:
i'm not sure what should be the expected behavior in such a case: i'd actually think that if disconnected / reconnect then the prompt should directly return no?

ThunderKing54, 2020-03-27:
https://www.youtube.com/watch?v=pC4PgV6WheY
I hope this explain it

limpkin, 2020-03-27:
but after disconnect, the orientation shown should be the one stored for when battery powered no?

commit 8c663537 (2020-03-27): fix #93
closed 2020-03-27 by limpkin