mooltipass/minible #94: battery "empty" when unplugging the mooltipass

ThunderKing54, 2020-03-26:
I found another battery bug ;D
### Expected behavior
If you disconnect your mooltipass more than twenty times in less then 7 seconds it should show the right battery value.
### Actual behavior
If you disconnect your mooltipass multiple times very fast the batteryindicator shows that the battery is empty
### Step by step guide to reproduce the problem
1) plug and unplug your mooltipass with usb over twenty times
2) insert smartcard and enter pin
3) the batteryindicator now shows a empty battery

### Firmware Version
AUX MCU version:  0.17
Main MCU version:  0.21

### Operating System
Mention if you are using either:
- Windows 10 Pro

limpkin, 2020-03-26:
twenty times xD 

looking into it :), nice find!

limpkin, 2020-03-26:
when you do this, is your battery initially full or discharged?

ThunderKing54, 2020-03-26:
It is full and I do it pretty fast.

limpkin, 2020-03-27:
[bundle.zip](https://github.com/mooltipass/minible/files/4396071/bundle.zip)

Could you try this firmware and let me know if that helps? I can't reproduce the bug so it makes things a bit tricky...

ThunderKing54, 2020-03-27:
I test it a bit

limpkin, 2020-05-03:
is this issue fixed? :)

ThunderKing54, 2020-05-04:
I think it is :). Thank you 👍

closed 2020-05-04 by ThunderKing54