mooltipass/minible #96: Animation freeze after 3 seconds during card cloning process

ThunderKing54, 2020-04-02:
:)
### Expected behavior
The animation for "Please Remove The Card" should not freeze after three seconds.
### Actual behavior
The animation for "Please Remove The Card" freeze after three seconds.
### Step by step guide to reproduce the problem
1) insert your smartcard and pin
2) go to the "Clone Card" menu
3) enter your pin
4) input new card pin
5) confirm pin
6) the message "Please Remove The Card" appear
7) wait some seconds. The animation start to freeze
8) if you unplug the mooltipass the screen goes black
9) same happens with "Insert New Card"

### Firmware Version
AUX MCU version:  0.18
Main MCU version:  0.22

### Operating System
- Windows 10 Pro

limpkin, 2020-04-02:
Thanks a lot for that issue!
I simply had forgotten to continue animating and calling power detection functions

closed 2020-04-02 by limpkin

ThunderKing54, 2020-04-03:
🤣 👍

commit 008c2c15 (2020-04-02): fix #96