mooltipass/minible #100: double selected output when bluetooth is enabled

ThunderKing54, 2020-04-03:
Something fancy again ;D

### Expected behavior
You should not be able to select two output interfaces
### Actual behavior
I'm able to select two output interfaces if bluetooth is enabled ;D
### Step by step guide to reproduce the problem
1) enter your smartcard and your pin
2) enable bluetooth and pair your mooltipass with your pc
3) connect your mooltipass with usb
4) go to the login menu and select a credential
5) The message "Select Output Interface" appear
6) now the tricky part start... Unplug the usb cable and scroll a tiny bit. If you scroll too long the bug won't be visible. If you scroll to late the bug won't appear. You must have the perfect timeing. 
### Firmware Version
AUX MCU version:  0.18
Main MCU version:  0.22

### Operating System
Mention if you are using either:
- Windows 10 Pro

[image]

ThunderKing54, 2020-04-03:
Oh excuse me it is much easier. 

1) attach the usb cable 
2) go to the enable bluetooth menu
3) Now it ask if you want to enable bluetooth
4) wait until the animation for yes start.
5) switch to no and unplug the usb cable
6) the selected choice goes back to yes
7) both items are now selected

limpkin, 2020-04-05:
As always, it's always a pleasure to fix your bugs :)

commit ba856f48 (2020-04-05): fix #100
closed 2020-04-05 by limpkin