mooltipass/minible #103: Webauthn adding new key for service bug

ThunderKing54, 2020-04-07:
### Expected behavior
If you disconnect your mooltipass during the webauthn authentication(adding new key) it should stop if bluetooth is not enabled.
### Actual behavior
If you disconnect your mooltipass during the webauthn authentication(adding new key) you can save the "key" and nothing is displayed on the website. If you try to add a key again nothing happens on the mooltipass.
### Step by step guide to reproduce the problem
1) connect your mooltipass with usb
2) insert your smartcard and pin
3) go to https://demo.yubico.com/webauthn and create a demo account
4) add a new key
5) wait 5 seconds after you get the message if you want to add the "credential"
6) unplug the mooltipass and accept the credential
7) when you repeat step 4 nothing happens on the mooltipass

### Firmware Version
AUX MCU version:  0.18
Main MCU version:  0.22

### Operating System
- Windows 10 Pro

limpkin, 2020-04-07:
@0x0ptr : i can have a look as well, up to you :)

limpkin, 2020-04-08:
as always, thanks for these awesome bugs!

commit 5a88223f (2020-04-08): fix #103
closed 2020-04-08 by limpkin