mooltipass/minible #104: Webauthn crashes Mooltipass Mini BLE

ThunderKing54, 2020-04-07:
### Expected behavior
If you disconnect your mooltipass during the webauthn authentication(adding new key) nothing should happen.
### Actual behavior
If you disconnect your mooltipass during the webauthn authentication(adding new key) the mooltipass crashes right after that.
### Step by step guide to reproduce the problem
1) connect your mooltipass with usb
2) insert your smartcard and pin
3) go to https://demo.yubico.com/webauthn and create a demo account
4) add a new key
6) unplug the mooltipass right after that
7) You can't do anything anymore. 

### Firmware Version
AUX MCU version:  0.18
Main MCU version:  0.22

### Operating System
- Windows 10 Pro

limpkin, 2020-04-07:
just to be clear... at what point should I disconnect the unit? during the prompt?

ThunderKing54, 2020-04-08:
right when the prompt appear

limpkin, 2020-04-08:
prompt on the device, correct?

ThunderKing54, 2020-04-08:
On the "Add New Credential" prompt on the device. Don't press anything. Unplug the usb cable right after the prompt is displayed.

limpkin, 2020-04-08:
holly molly that was a tough one that required me to change the internal logic... I'll have to send you a new firmware for you to continue your testing :D

commit 4227a023 (2020-04-08): fix #104
closed 2020-04-08 by limpkin