mooltipass/minible #105: Abort sleep mode during credentials recall

ThunderKing54, 2020-04-08:
Don't know if it is a bug or a feature :)
### Missing feature
Abort sleep mode when visiting a website with saved credentials

### Justification
If I visit a website it takes about two seconds until I'm able to recall the credentials if the "Going To Sleep" animation starts.

### Workarounds
Keep the Mooltipass sleepless

### Step by Step guide
1) connect with usb
2) enter smartcard and pin
3) open a new website with saved credentials
4) cancel the credential recall on the mooltipass
5) wait until the "Going To Sleep" animation start
6) Press F5 and reload the website
7) nothing happen until the Mooltipass is in "Sleep" mode. After that the Mooltipass display a message if I want to send the credentials.

limpkin, 2020-04-08:
[bundle.zip](https://github.com/mooltipass/minible/files/4452586/bundle.zip)

Thanks a lot @ThunderKing54 !

Attached is a new bundle update for the 3 bugs that I just fixed... could you continue your testing with it? :)

Again.... thanks a lot :)

closed 2020-04-08 by limpkin

ThunderKing54, 2020-04-11:
Thank you @limpkin :)

commit 16c3e50f (2020-04-08): fix #105