mooltipass/minible #109: No password and username output on iOS with bluetooth

ThunderKing54, 2020-04-21:
Even tested it with Keyboard Layout for MacOS(Germany) without success!
### Expected behavior
If you pair your mooltipass with your Iphone the mooltipass should be able to send the credential
### Actual behavior
If you pair your mooltipass with your Iphone nothing happens if you select a credentials in the "Login" menu
### Step by step guide to reproduce the problem
1) enter smartcard and pin
2) enable bluetooth on Iphone and mooltipass
3) connect your mooltipass with your Iphone
4) go to any website and select the credentials inside the "Login" menu
5) A prompt appear if you want to type in the Login and password
6) I accepted both prompts but nothing happen

### Firmware Version
AUX MCU version:  0.20
Main MCU version:  0.24

### Operating System
- Iphone 8 plus iOS 13.3.1

ThunderKing54, 2020-04-22:
If the Mooltipass is connected with my Iphone I'm not able to use the on screen keyboard anymore 🤣 It is gone 🧐

limpkin, 2020-05-17:
@ThunderKing54 there's an option to force the virtual keybard display IIRC

limpkin, 2020-05-17:
For reference:
- https://github.com/asterics/esp32_mouse_keyboard/issues/4
- https://github.com/asterics/esp32_mouse_keyboard/commit/efed298d69e99d4721feb02822afbad39714a12c
- https://github.com/asterics/esp32_mouse_keyboard/commit/efed298d69e99d4721feb02822afbad39714a12c

commit b85fb7d8 (2020-06-05): fix #109
closed 2020-06-05 by limpkin