mooltipass/minible #110: security key and Windows hello bug

ThunderKing54, 2020-04-24:
### Expected behavior
If you want to delete/reset the security key under windows it should work without any problems
### Actual behavior
If you want to delete/reset the secuirty key under windows, windows says that it could not reset the secuirty key.
### Step by step guide to reproduce the problem
1) connect your mooltipass with usb
2) enter your smartcard and pin
3) go to preferences > account >sign-in options > security key
4) add your security key
5) go back and püress manage
6) accept the promp on the mooltipass to update the data for "Selected Device"
7) Press the button "reset/zurücksetzen"
8) press okay and accept the promp on the mooltipass

### Firmware Version
AUX MCU version:  0.20
Main MCU version:  0.24

### Operating System
Mention if you are using either:
- Windows 10 Pro 18362

[image]
[image]

limpkin, 2020-04-24:
@0x0ptr are you free to give it a try? :)

0x0ptr, 2020-08-18:
We have decided to not allow Reset of the FIDO2 credentials using this feature. The "Reset" feature of FIDO2 is meant to factory reset the device. One probably don't want that since the Mooltipass is storing other credentials too. Managing the device (including deleting/resetting FIDO2 credentials) should be done from one place which is the Moolticute app.

ThunderKing54, 2020-08-18:
Oh okay :). Good to know. I thought it only want to delete all keys for windows and failed.

0x0ptr, 2020-08-31:
https://github.com/mooltipass/minible/pull/177

closed 2020-09-01 by 0x0ptr

ubergeek77, 2022-07-17:
@limpkin Apologies for the ping for such an old issue, but I thought it would be cleaner to keep Windows Hello stuff here.

I'm trying to *add* my Mini BLE as a sign in method for Windows, but I'm not getting the option to set it up in the first place. When I plug in the Mini BLE, all I am presented with is this:

[image]

Notice how the only option is to "Reset" - there is no option to set up a PIN so that this can be used as a login method. My Mini BLE has never been configured as a security key for Windows, and my Windows device has never had *any* USB security keys configured before. So I'm not sure why Windows is treating this like an already-configured USB key.

It seems like the expected flow is that I should first see a "Touch your security key" prompt, as seen in this video:

https://youtu.be/LRjCXZCCkvE?t=9

However, with the Mini BLE plugged in and unlocked, I do not get this prompt. With my device, Windows skips straight to "Reset your security key." I never get the option to set a PIN or use the Mini BLE as a sign in method.

It seems Windows is at least detecting the presence of the Mini BLE though, because with the device unplugged, I see this prompt:

[image]

But, I still can't figure out how to get this set up. I don't know if this is a bug, or if I am just doing something wrong here.

If you have any advice, it would be greatly appreciated :)
 
(On Bundle 8)

limpkin, 2022-07-17:
@0x0ptr hello! how are you doing :).
@ubergeek77 : is this with microsoft.com work or personnal account? see https://github.com/mooltipass/minible/wiki/FIDO2-Site-Compatibility-Chart

ubergeek77, 2022-07-17:
Ah, I wasn't aware of the chart.

I didn't know USB security key login was tied to a Microsoft account either, Microsoft doesn't make this clear in the UI at all.

I tried this on two Windows machines - one without a Microsoft account, and one that is set up with a work account. I saw the same behavior on both, and I had no idea they were failing for different reasons.

Ultimately, I was only really interested in getting it working on my work account machine, and trying it on my Microsoft account-less device was just a troubleshooting measure. But, seeing your compatibility chart, it looks like your hands are tied for now due to Microsoft's requirements.

Are these hardware requirements? Or will this perhaps be supported for work accounts in the future?

Anyway, thanks for letting me know about the compatibility!

limpkin, 2022-07-17:
oh sorry :/
these aren't hardware requirements... hopefully someone can get to implementing the missing features :)