mooltipass/minible #111: Mooltipass shows as "connected" but it doesn't allow entering credentials via Bluetooth.

GlitchOne, 2020-04-24:
### Expected behavior
Being able to enter credentials when the device displays as "connected" in Windows.

### Actual behavior
Device does not seem to actually be connected.
Device only allows the option to "display" credentials.

### Step by step guide to reproduce the problem

1. Pair Mooltipass with a Windows Laptop.
2. Disconnect Mooltipass from Windows using the "Disconnect device" option.
3. Pair device with an Apple iPad.
4. Disconnect Mooltipass from Apple iPad using the "Disconnect device" option.

After doing that, the Mooltipass will show as connected on the Windows machine, but the user will not be able to enter credentials. 

Stopping and starting the Bluetooth functionality will connect the device to the Windows machine and allow the user to enter credentials via Bluetooth.

### Moolticute Version - If Involved
N/A

### Firmware Version
AUX MCU version: v0.24
Main MCU version: v0.20

### Operating Systems
- Windows 10 Enterprise, v1909, Build 18363.752 
- iOS 13.3.1

limpkin, 2020-04-30:
I can't reproduce the issue with an android phone... do you have an android phone to test this bug as well?
Just trying to understand if it is ios related...

GlitchOne, 2020-04-30:
I do not have an Android device that can connect to the Mini BLE, unfortunately. I can, however, try multiple different combinations, to see if there are other root causes. Will be back with updates as soon as I have them.

limpkin, 2020-04-30:
that would be awesome, thanks :)

GlitchOne, 2020-05-01:
I am back with some additional info:

1. I tried using two Windows 10 machines for testing pairing combinations. One of them uses a Trust Bluetooth dongle, the other is a laptop with built-in Bluetooth. Sometimes (and I have not been successful in defining clear steps to reproduce this) the issue appears between these two Windows machines as well. What I noticed is that whenever typing credentials isn't available, the Windows machine will show the device as connected but the Bluetooth icon on the Mini BLE will not have those little triangles to the left and right of it.
2. Additionally, after about a minute, the Mini BLE will reconnect to the device it was previously connected to.

I have managed to apparently reproduce this issue first thing after I updated to the newest firmware.

1. I updated the firmware.
2. Removed the pairings from the windows machines.
3. Removed the pairings from the Mini BLE.
4. Connected to the dongle equipped desktop.
5. Disconnected.
6. Connected to the laptop.
7. Disconnected.

At that point, the Mini BLE was showing as connected on the Dongle equipped Desktop but without it actually *being* connected. After about a minute, it successfully reconnected to the laptop.

I will try and see if I can reproduce this again somehow. Is there any way of restoring the Mini BLE to some kind of factory-state so that I know for a fact I have a clean slate to work off of?

limpkin, 2020-05-01:
Thanks for the time spent on this!
Just to make sure I follow you... you're saying it still shows as connected after you select the disconnect item?
As far as BLE goes, removing pairings on the mini and the computers is enough!

GlitchOne, 2020-05-01:
No, after I disconnect from the laptop, it shows as if it's connected to the desktop, even though it's not, really.

limpkin, 2020-05-05:
so, re-reading this issue I should explain a bit more what the mini does when you select the disconnect button: in short, it temporarily "bans" the currently connect device for 30 seconds.
The behavior you describe at step 7 is therefore expected: the ble should reconnect to the dongle equipped desktop. I'm therefore wondering if something fishing is happening with your dongle equipped desktop. could you maybe try to enable/disable bluetooth at step 6 on the dongle equipped laptop and see if you notice any change?

GlitchOne, 2020-05-05:
I will try that tomorrow together with some other use cases and get back to you with some updates.

limpkin, 2020-05-07:
@GlitchOne any updates? :)

GlitchOne, 2020-05-08:
Yes, sorry, work has been hectic lately. 

Before starting, I have to mention that I'm using the v0.26/v0.21 bundle for the tests now.

1. The iPad issue still replicates as per the initial reproduction steps.
2. Disabling Bluetooth on the machines do not seem to affect the issue, but I will try again after my Mini BLE finishes charging.
3. It would seem that the issue happens on Windows machines at least once after an initial connection to the iPad.

These things lead me to believe that maybe the Bluetooth communication with Apple devices makes the Mini BLE act a bit weird. I will try to get my hand on an iPhone as well and see whether or not Apple/Apple connections are also affected somehow. Will update in about seven hours.

GlitchOne, 2020-05-09:
So it's been more than seven hours, but alas, I have used two Apple devices and doing the same reproduction steps as the initial issue does not show any problems. So it's _maybe_ an Apple/Windows issue, somehow?

limpkin, 2020-05-17:
Given the recent fw update, i'll close it... feel free to open a new one :)

closed 2020-05-17 by limpkin