mooltipass/minible #112: too short underline headline inside keyboard layout for bluetooth

ThunderKing54, 2020-04-27:
### Missing feature
Calculate the length of the underline depending on the with of the headline. If you go into the keyboard layout menu for bluetooth the underline is too short. 

### Justification
It clearify the headline

### Workarounds
there are no workaournd at the moment

limpkin, 2020-04-27:
which underline are you talking about? :) do you have a picture? not sure to follow

ThunderKing54, 2020-04-27:
[image]
[image]
[image]

ThunderKing54, 2020-04-27:
"Selected Service" is fine(last picture)
But the underline for "Select USB Layout" and "Select Bluetooth Layout" is too short for my opinion. The underline length should by calculated by the headline text length

limpkin, 2020-04-27:
Ohhhh that's a nice touch.

ThunderKing54, 2020-04-27:
😄

limpkin, 2020-04-29:
fixed! also fixed that "bluetoth" typo
thanks a lot :)

closed 2020-04-29 by limpkin

ThunderKing54, 2020-04-30:
Didn't noticed the "bluetoth" typo 😆. It is the tipical "The Significance of Letter Position in Word Reconition"

commit 410357b9 (2020-04-29): fix #112