mooltipass/minible #113: Slowly click-scrolling credentials hard-resets the device

Arffeh, 2020-04-28:
### Expected behavior
While actively clicking down the scroll-wheel and scrolling successfully through the login credentials, 5-second hard-reset timer shouldn't be counting.

### Actual behavior
Timer triggers while click-scrolling through credentials, If I'm too slow and the 5 second timer expires, the MP-BLE hard resets.

### Step by step guide to reproduce the problem
  - Enter PIN
  - Select Login
  - Hold down scroll wheel
  - Start slowly scrolling through your lovely massive set of credentials
  - After5 seconds the device will trigger a reset from holding down the scroll wheel too long.

### Firmware Version
AUX MCU version:  0.20
Main MCU version:  0.24

### Operating System
Arch Linux (not relevant)

limpkin, 2020-04-28:
It's actually a 10 seconds timer... perhaps I should extend it to 15 seconds as pushing without scrolling for 10 seconds straight could get tricky...

Arffeh, 2020-04-28:
If it's 10, then I must be having a slightly different problem. 
I just used a stopwatch, it happens at 6 seconds. Tested multiple times.

6 is oddly specific.

limpkin, 2020-04-29:
You're indeed perfectly right!
https://github.com/mooltipass/minible/blob/master/source_code/main_mcu/src/platform_defines.h#L422

Will increase it to 10 and mark it fixed :), feel free to reopen if you feel that it is still to short :)

commit bac94771 (2020-04-29): fix #113
closed 2020-04-29 by limpkin