mooltipass/minible #114: Empty Battery Indicator woth Full Battery

dehne, 2020-04-28:
### Expected behavior
Full battery indicator showing after overnight charge
### Actual behavior
Empty battery indicator showing until forced reboot
### Step by step guide to reproduce the problem
1. Charge the device normally until the battery icon shows a full battery. 
2. Unlock it and enable Bluetooth. Connect to a device, my Android phone, in this case. (Not sure this is necessary, but that's what I did.) 
3. Carry the device around until waking it up leads it to complain that its battery is depleted. (In my case, that was only about 15 hours (?!))
4. Put the device on charge overnight. In the morning, wake it up. Instead of a full battery icon, the empty battery icon (the one with the "X" over the battery) displays.
5. Long press the scroll wheel until the screen goes black.
6. Long press the scroll wheel until it boots.
7. Unlock the device. The battery icon shows "full battery."
### Moolticute Version - If Involved
N/A
### Firmware Version
AUX MCU version:  0.20
Main MCU version:  0.24

### Operating System
N/A
### Mooltipass Extension
N/A

limpkin, 2020-04-29:
I'm fairly sure I fixed it here: https://github.com/mooltipass/minible/commit/d0ca92d7bfc4556d81fe137587bfd448404763ac#diff-d71195a827299e80e909648e69207aa6L33

Will leave it opened until you get to test this :)

dehne, 2020-05-01:
I was able to reproduce this a second time following the instructions in the report. 
[image]
It may be that not waking the device up immediately after beginning to charge the battery is important to reproducing the problem. In any case, I'll update to the 0.21/0.25 firmware and test again.

limpkin, 2020-05-01:
that's some nice card cutout you have there!
Yep, please update and let me know if that helped :)

limpkin, 2020-05-03:
@dehne looking forward to hear if the latest update worked :)

dehne, 2020-05-04:
Me too. I can tell you this: the battery optimizations have certainly worked. Even with Bluetooth enabled, it's taking a long time to run down the battery to empty! It's still going strong after two days running on battery (except for a few minutes plugged in to update the firmware).

limpkin, 2020-05-04:
Woohoo! the hard work is starting to pay off!

limpkin, 2020-05-06:
@dehne how's everything going? :)

dehne, 2020-05-06:
It's been running for five days, two with a single bar on the battery and now three showing no bars at all, but still going strong. Just waiting for it to say the battery is empty so I can do the test.

limpkin, 2020-05-07:
5 days is what I was shooting for! I'm very happy :)

dehne, 2020-05-09:
This morning, the device finally reported an empty battery. After it shut itself down, I plugged it in and let it charge for a while, When I then woke it up, it did NOT show the battery icon with an X on it. So using the latest firmware, I can no longer reproduce the problem. Time to close this and move on.

closed 2020-05-09 by dehne

limpkin, 2020-05-09:
woohoo!