mooltipass/minible #117: No Text Output Over Bluetooth on Nokia 7.2/Android 9

ghalfacree, 2020-05-01:
I've been mentioning this on the mailing list, but now we're on the RC firmware figured it needed extra attention - apologies if it duplicates a closed issue, but it's definitely still a problem here!

### Expected behavior
When connected to a Nokia 7.2 Android 9 smartphone, the BLE acts as a Bluetooth keyboard and allows manually-recalled credentials to be "typed".

### Actual behavior
Nothing. Nada. Zip.

### Step by step guide to reproduce the problem
Pair BLE with Nokia 7.2. Check it is registered as an input device. Open anything that accepts a text input. Recall a credential on the BLE. Confirm you want the username typed.

### Moolticute Version - If Involved
N/A

### Firmware Version
AUX MCU version:  0.21
Main MCU version:  0.25

### Operating System
Android 9

limpkin, 2020-05-17:
@ghalfacree looking forward to hear from you after this last fw update :)

ghalfacree, 2020-05-18:
@limpkin Upgraded to 0.30/0.24, and as per the mailing list: it's now fixed. Huzzah!

closed 2020-05-18 by ghalfacree