mooltipass/minible #118: Small UX Issue: Scroll wheel icon on the left side of the screen when in right-handed mode.

GlitchOne, 2020-05-01:
### Missing feature
The scroll wheel icon in the tutorial is on the left side of the screen when the Mini BLE is in right-handed mode.

### Justification
I believe it would be more intuitive to have it on the same side as the actual scroll-wheel, for a better experience when first doing the tutorial.

### Workarounds
N/A

limpkin, 2020-05-01:
I actually wondered about that..... I'm still torn... as on the screen we're displaying what the wheel looks like viewed from the user no?

GlitchOne, 2020-05-01:
Well, yes. So when the mooltipass is in right-handed mode, the scroll-wheel is on the right and I would assume that the icon in the tutorial would be on the right as well. But when it's in left-handed mode, I would assume that the icon stays on the left, the same side as the scroll-wheel.

Don't get me wrong, it's not a huge issue by any means, just a nice-to-have in my opinion.

limpkin, 2020-05-01:
I indeed meant that :).
As I said, I'm still fairly torn.... we'll see what other users think :)

ThunderKing54, 2020-05-03:
I'm with GlitchOne. It is more intuitive when the scroll wheel is on the right side when you use the right-handed mode. Isn't the Mooltipass Mini Tutorial like that?

kbembedded, 2021-06-16:
FWIW I agree with the original intent. Especially considering there is an animation to prompt the user that the scroll wheel can be pressed.

If you want to clean up this tutorial experience, you might consider adding to the animation the edge of the device to kind of hint more to the user that the screen is showing the right edge of the device with the scroll wheel.