mooltipass/minible #120: disconnect should alternate between all devices

elendil-software, 2020-05-01:
Use one of the two templates below and delete the other. All issues not following this template will be directly closed.

8<------------------------ BUG REPORT -----------------------------------------
### Expected behavior

When using several device, disconnecting devices should reconnect to next device, then the next, etc.

### Actual behavior

Seems to work correctly with two devices, but when using more device it keep switching between two same devices

### Step by step guide to reproduce the problem

- Enable Bluetooth on one of the device.
- Connect the Mooltipass to this device
- Enble Bluetooth on the two other devices
- Disconnect the Moolticute (menu Bluetooth -> Disconnect device)
- Mooltipass will connect to device 2
- Disconnect again
- Mooltipass will connect to device 3
- Disconnect again.
- Mooltipass will connect to device 2
- etc.

From this point it continue to alternate between device 2 and 3 and never reconnect to device 1 unless bluetooth is disabled on device 2 and 3.

### Moolticute Version

v0.43.14-testing

### Firmware Version
AUX MCU version: 0.21
Main MCU version: 0.25

### Operating System
  
Other operating systems are not supported.

### Mooltipass Extension
- If you're creating an issue to report a website incompatibility, please use the "Report incompatibilities with this website" item in the extension menu.
- If you're creating an issue to report a problem with our extension, please create another issue here: 
https://github.com/mooltipass/extension/issues

8<------------------------ END BUG REPORT -------------------------------------

8<------------------------ FEATURE REQUEST ------------------------------------
### Missing feature

### Justification
Tell us why you would like to see this feature added.

### Workarounds
Are there any workarounds you currently have in place because the feature is missing?

8<------------------------ END FEATURE REQUEST --------------------------------

limpkin, 2020-05-02:
That's going to be a tough one, as it isn't possible for the device to know if device 1 is still there....  
do you really have that use case?

WebmasterTD, 2020-05-03:
Maybe there could be a list of paired devices and choose from there manually witch one we want to switch.

elendil-software, 2020-05-03:
> 
> 
> That's going to be a tough one, as it isn't possible for the device to know if device 1 is still there....

Ok, but then, when connected with device 2, how does it know device 3 is there ?

> do you really have that use case?

With three PC, no. But it's not uncommon I have my PC, laptop and my phone with bluetooth enabled

elendil-software, 2020-05-03:
> 
> 
> Maybe there could be a list of paired devices and choose from there manually witch one we want to switch.

Or the possibility to force the connection from the OS Bluwtooth control panel.

limpkin, 2020-05-03:
The way it is currently implement is a temporary ban on the currently connected device.  
Choosing from a list could indeed be an option...

limpkin, 2021-04-27:
closing this one as 150 is a better suggestion

closed 2021-04-27 by limpkin

elendil-software, 2021-04-28:
> closing this one as 150 is a better suggestion

👍