mooltipass/minible #121: Service name touches/covers animation

ThunderKing54, 2020-05-05:
### Expected behavior
If a websites name is longer than 27 characters it should not cover or touch the animation to accept or deny the credential
### Actual behavior
If a websites name is longer than 27 characters it cover or touch the animation to accept or deny the credential
### Step by step guide to reproduce the problem
1) enter your smartcard and pin
2) got to the "Login" menu
3) Select a Service which have 27 and more characters
4) You see the prompt if the Mooltipass should type the Login
5) The Service name plus username touches the animation to accept or deny the credential

### Firmware Version
AUX MCU version:  0.21
Main MCU version:  0.26

### Operating System
Mention if you are using either:
- Windows 10 Pro

[image]

limpkin, 2020-05-05:
I like it this way actually.... if I'd truncate a bit more to the left it'd look a bit odd no?

ThunderKing54, 2020-05-05:
Scroll your scrollwheel during the animation. You gonna see that the text is cutted of. Maybe if you're able to fix that it would be fine as well.

limpkin, 2020-05-05:
[bundle.zip](https://github.com/mooltipass/minible/files/4583250/bundle.zip)
@ThunderKing54 could you try the attached bundle and let me know what you think? it's a little more slower and therefore less prone to mistakes :)

closed 2020-05-05 by limpkin

ThunderKing54, 2020-05-05:
@limpkin awesome! :) I like it very much but I don't know if other people like it. It happen quite often to me that I scrolled too far and denied adding a new credential or accepted a credential by mistake. I don't know if other people like the slow animation and reaction but we will see the response of the other beta testers. Thank you very much @limpkin. If other people don't like it, could you add a new setting to change between slow reaction and fast reaction in Moolticute?

limpkin, 2020-05-05:
Glad you like it! Let's see what the other testers think as well :).
Not sure if I want to make it a setting though :)

commit dcefc157 (2020-05-05): fix #121