mooltipass/minible #123: Credential Mirror Feature

limpkin, 2020-05-07:
From a beta tester request:

I would like to be able to link multiple services or add multiple
domains for a single credential. Some web services use different domains
with one login. For example facebook.com - messenger.com, alibaba.com -
aliexpress.com
Would be convenient to have it under one credential so if i change my
password, I won't forget to change in multiple places.

hijackeel, 2021-06-18:
It would also be neat if for the site name, you could enter something like "sitename.*" or perhaps more securely "sitename.[com,net,co.uk,fr,de,ch]" if multiple TLDs for the same site are possible.

bertvandepoel, 2021-07-28:
A specific use case for this: the Belgian government hosts an IdP for government stuff on idp.iamfas.belgium.be but if you want to administer your login methods, keys and tokens, you go to iamapps.belgium.be. However, the scope of belgium.be is much too wide, so for something like this, a way to link things would be nice indeed!

limpkin, 2022-05-13:
@bertvandepoel @paquetj @WebmasterTD @rms5566 @elendil-software @kbembedded @Properrr @hijackeel @wasserpanther : instead of having to manually add mirrored credentials, a friend of mine had the idea to add a "wildcard" feature. 
The general idea would be to add a boolean for each credential, that if set would allow the device to answer credentials requests to domain.XYZ

That obviously only works if the other service is on the same domain and is very similar to @hijackeel idea. @bertvandepoel belgium.be example could be addressed by adding a feature on moolticute's side of things.

WebmasterTD, 2022-05-13:
I think this solution could be useful in some cases. However according to Apple: [shared credentials](https://github.com/apple/password-manager-resources/blob/main/quirks/shared-credentials-historical.json) a lot of sites share completely different domains.

limpkin, 2022-11-05:
this is now implemented in the latest bundle v12 :)

closed 2022-11-05 by limpkin

elendil-software, 2022-11-27:
How has it been implemented ? Where it has to be enabled ?

limpkin, 2022-11-27:
bundle v12, have a look at our user manual :)