mooltipass/minible #124: Freezing over BLE

WebmasterTD, 2020-05-08:
###  Behavior
When the device is connected to my PC through BLE it sometimes freezes and locks up. Not responding to anything. I have to remove the card and try to reset it with the click wheel a few times to wake it up. I can´t really figure out what causes this behavior, so i can't give the steps to reproduce this issue. However this happened to me multiple times.
Not sure if this is related to this but once when i tried to wake up the device it displayed - No Aux MCU.

### Moolticute Version
0.43.15

### Firmware Version
AUX MCU version:  0.21
Main MCU version:  0.26

### Operating System
- Windows 10

limpkin, 2020-05-08:
I'm definitely trying to find what's the root cause of that bug... but can't reproduce it....
I would really appreciate if you could find a way to reliable replicate that issue :)

WebmasterTD, 2020-05-09:
I got debug logs before the device froze
`WARNING: (2020-05-09T11:46:22.152) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.154) :0 - Full response:  "3e080000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.167) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.168) :0 - Full response:  "3e180000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.183) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.184) :0 - Full response:  "3e280000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.186) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.186) :0 - Full response:  "3e380000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.197) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.197) :0 - Full response:  "3e480000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.200) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.200) :0 - Full response:  "3e580000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.212) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.212) :0 - Full response:  "3e680000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.228) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.228) :0 - Full response:  "3e780000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"
WARNING: (2020-05-09T11:46:22.230) :0 - Wrong answer received:  "EXPORT_FLASH_START (0x008a)"  for command:  "GET_CREDENTIAL (0x0007)"
WARNING: (2020-05-09T11:46:22.230) :0 - Full response:  "10880000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"`

limpkin, 2020-05-10:
nice!
did you get what was before that?

WebmasterTD, 2020-05-10:
I don´t have those logs anymore :(
Next time I will include those as well. 
Question: Debug logs are safe to share? Does not include passwords or secret information?

limpkin, 2020-05-11:
Debug logs: ask me in private before sharing :), as there are some messages exposing passwords indeed

crossing my fingers for you to get another freeze :)

limpkin, 2020-05-17:
hopefully this is fixed with the latest fw release!
will close in a week if you don't experience this again.

elendil-software, 2020-05-19:
Hi,

I was on the point of opening a similar issue. I encounter the same kind of issues. But so far I didn't managed to find a way to reproduce it every time.
One thing I noticed, in my case, is that it seems to happens when Bluetooth is interrupted on the computer side (on power of for example)

I will try with the last firmware.

limpkin, 2020-05-21:
About to release one that could fix that issue! v0.31

elendil-software, 2020-05-23:
OK, I hope I don't speak to fast but I think it's solved. I will do more test, but so far the previous behaviour didn't occurred again.

limpkin, 2020-05-23:
don't give me false hopes :p 

anyway, that was a tricky one to find :)

limpkin, 2020-05-24:
no news.... is it still looking good @elendil-software @WebmasterTD ?

elendil-software, 2020-05-24:
Concerning my specific case (freeze after bluetooth on computer), I think it's solved. I tried to reproduce the problem the last two day but it never freezed again.

But I think I found a new case that occur when paring has been deleted on the Mooltipass but not on the computer. The problem doesn't occur every time but here is what I've done the three times it froze : 
1. Delete pairing from Mooltipass
2. Start a computer (didn't try with other device than windows 10 computers) that has been previously paired
3. Sometimes the Mooltipass freeze

limpkin, 2020-05-24:
I can't reproduce the bug.... how do you know that the mooltipass is frozen? a single click doesn't wake it up?
The paired to connected toggling is due to windows and normal.

limpkin, 2020-05-24:
I however noticed that in this case won't advertise... will fix this now :)

elendil-software, 2020-05-24:
> I can't reproduce the bug.... how do you know that the mooltipass is frozen? a single click doesn't wake it up?

Exactly. 

> The paired to connected toggling is due to windows and normal.

OK.

WebmasterTD, 2020-05-24:
I haven't experienced the bug since last update.

limpkin, 2020-05-24:
thanks a lot @elendil-software !!!!! your way to reproduce your issue allowed to find an issue that could cause sporadic freezes..... this is truly awesome!

Expect a nice firmware update in the coming days :)

closed 2020-05-24 by limpkin

elendil-software, 2020-05-25:
Great !
I hope it was the last of this kind of "not every time freeze" bug. Not easy to reproduce. :-)

Julien