mooltipass/minible #126: Unable to "type" via BLE on first connection to MacOS

kevin-sinclair, 2020-05-11:
------------------------ BUG REPORT -----------------------------------------
### Expected behavior

After having disconnected the Mooltipass BLE from USB and deleted all pairings, you successfully pair the device with a Mac and you can start "typing" passwords over BLE.

### Actual behavior

After having disconnected the Mooltipass BLE from USB and deleted all pairings, you successfully pair the device with a Mac and you are unable to "type" passwords over BLE.
Switching Bluetooth off and then on from the device, reconnects it with the Mac and the "typing" of password is now possible.

### Step by step guide to reproduce the problem
1. Mooltipass is disconnected from USB and Pairings have been cleared  from the "Delete Pairings" menu;
2. Pair successfully Mooltipass BLE with MacOS (Mojave);
3. Open a blank text file
4. "Type" a username/password via the Mooltipass "Login" menu;
5. Nothing gets "typed" into the text file.

### Moolticute Version

N/A

### Operating System

- MacOS Mojave 10.14.6
  
### Mooltipass Extension

N/A

### Mooltipass Device
- The Mooltipass Mini BLE

------------------------ END BUG REPORT -------------------------------------

limpkin, 2020-05-11:
yeah.... that's.... something.

limpkin, 2020-05-17:
So.... after digging through the console logs, I have noticed an abnormal behavior of MacOS when pairing for the first time a BLE device containing two HID primary services.
Upon successful pairing, macOS will try to create two HID devices with the SAME keyboard report map, instead of creating two each HID devices, each with their own report maps. 
Upon disconnecting & reconnecting the device, the correct behavior is seen.
By looking at the attached screen captures, you'll notice the behavior:
- attempt to create two HID devices with the same report map of length 45 (see popup and window below)
- attemps to create two HID devices, one with a report map of 45 bytes, the other with a report map of 65 bytes

Issue has been reported to Apple....

[image]
[image]

iaglet, 2020-05-22:
issue remains with v0.31, experienced with apple iphone 11 ios 13.4.1

limpkin, 2020-05-22:
the macos issue is unlikely to go away as it is an issue in MacOS itself.
We're however working on the iOS issue.

limpkin, 2020-06-06:
FYI iOS issue fixed!