mooltipass/minible #128: Add wheel "click shortcuts" for BLE actions

elendil-software, 2020-05-19:
### Missing feature

wheel "click shortcuts" for BLE actions

### Justification
Enabling/Disabling Bluetooth, switching from device to an other could be thing that could happens quite frequently. It would be nice to be able to do these action with some kind of wheel click shortcut. For example a double-click could execute the disconnect action (and therefore reconnect to other device). A triple-click could enable/disable bluetooth.

### Workarounds

Use the menu options

limpkin, 2020-10-05:
I'm thinking of these when in the main menu:
- wheel press - scroll up - scroll down (still pressed) to disconnect
- wheel press -scroll down - scroll up (still pressed) to enable / disable bluetooth

would that work?

elendil-software, 2020-10-05:
Yes, probably

elendil-software, 2021-02-26:
I think this issue can be closed.

limpkin, 2021-02-26:
sweet!

closed 2021-02-26 by limpkin