mooltipass/minible #131: Windows/Mooltipass report different connected status

elendil-software, 2020-05-22:
### Expected behavior

Connected status should be the same in Windows and on Mooltipass

### Actual behavior

On some occasions when I disconnect a device and it reconnect to an other device, the status shown in the Windows Bluetooth settings panel is connected but on the mooltipass it is shown as disconnected (confirmed by a message when trying to disconnect it again from the Mooltipass menu).

### Step by step guide to reproduce the problem

So far I have not precise steps to reproduce. But it seems to occur only when disconnecting from a device and reconnect to an other. And never when connecting to one device (after powering on computer for example)

### Moolticute Version

0.43.17

### Firmware Version
AUX MCU version: 0.24
Main MCU version: 0.31

### Operating System
- Windows 10 Pro

elendil-software, 2020-05-22:
After doing some more test I noticed that the status indicator in Windows some times change constantly between paired and connected

limpkin, 2020-09-05:
what's the status of that one @elendil-software ?

elendil-software, 2020-09-05:
I haven't noticed this problem since a long time. I think we should close it

closed 2020-09-05 by elendil-software