mooltipass/minible #132: Battery level indicator realibility

elendil-software, 2020-05-22:
### Expected behavior

Battery level show actual battery level

### Actual behavior

When I look at the battery level I'm not confident if what is shown is correct. 
Some examples :
- The level shown in the Windows Bluetooth panel does not match what is shown on the Mooltipass screen. I can have 25% on the Mooltipass and 90% on Windows. Looks like it's is not updated every time on Windows side.
- I had a notification that told me that battery level was 0% but have been able to use the Mooltipass intensively quite a long time. Don't know exactly how long but much longer than other devices I own which require to be charged shortly after the notification to be able to continue to use it. Maybe it's different with Mooltipass because it is not using so much energy.
- After having connected the Mooltipass to USB to recharge it from a almost empty battery, it took only 15-20 min to have the level around 100%. But 30-40 min later it's again around 25%

### Moolticute Version

0.43.17

### Firmware Version
AUX MCU version:  0.24
Main MCU version:  0.31

### Operating System

Windows 10 pro

limpkin, 2020-05-22:
So I'm seeing 2 issues here:
1) missing BLE notification about battery status
2) low battery performance: have you performed a reconditioning test? :)

elendil-software, 2020-05-22:
> low battery performance: have you performed a reconditioning followed by a battery test?

Yes, should I to try that again ?

limpkin, 2020-05-22:
maybe not actually... i'll come up with a better way to measure battery level :)

limpkin, 2020-09-05:
how is it doing @elendil-software  ? should we close that one?

elendil-software, 2020-09-05:
Much better. We can close this issue

closed 2020-09-05 by elendil-software