mooltipass/minible #135: Rename: File Management instead of Database Management

ThunderKing54, 2020-06-04:
### Missing feature
Rename the prompt from "Manage Database" -> "Manage Files" and "Database Management" -> "File Management" for the Files Management prompt and Information animation.
### Justification
You're able to see the differences in manage the Password database or the saved files

### Workarounds
no workaround available

limpkin, 2020-06-04:
Hmm that's a tricky one, as for the device itself there's no difference... don't think there's a real value to implement this....

ThunderKing54, 2020-06-04:
Ooh okay. Never change a running system! :) Didn't know that it is one system.

kevin-sinclair, 2020-12-14:
Or go for "Manage Data?" and call it a day?

kbembedded, 2021-06-16:
I understand the desire to differentiate, but I think clarifying documentation that all MMM is really just working with a database and that all of the options will prompt to manage database should suffice.