mooltipass/minible #136: _unlock_ service locks itself again after second unlock

ThunderKing54, 2020-06-04:
### Expected behavior
If you use the unlock Service it should not lock the Mooltipass again
### Actual behavior
If you use the unlock Service the Mooltipass locks itself every second try.
### Step by step guide to reproduce the problem
1) insert smartcard and pin
2) enable _unlock_ service with two accounts
3) set the unlock feature to Enter + Pass / WIN + L
4) unplug the smartcard
5) windows is locked and Mooltipass as well
6) enter smartcard and pin
7) windows is unlocked but the Mooltipass is locked again
8) The mooltipass goes in sleep mode and wake up again after less then 1 second and it won't stop. 
### Moolticute Version
v0.43.17-testing

### Firmware Version
AUX MCU version:  0.25
Main MCU version:  0.32

### Operating System
- Windows 10 Enterprise 2016 LTSB

limpkin, 2020-06-04:
ugh that's odd as i can't replicate this one....
could you try the attached bundle? it implements a fix that should help
[bundle.zip](https://github.com/mooltipass/minible/files/4731899/bundle.zip)

ThunderKing54, 2020-06-05:
I gonna try that one

ThunderKing54, 2020-06-05:
It is still there. I forgot to mention that I use the easy mode and not the advanced one

ThunderKing54, 2020-06-05:
Maybe the informations above are enough to reproduce it...

limpkin, 2020-06-05:
hmmm still not... is it through USB or BLE?

ThunderKing54, 2020-06-05:
1) connect your mooltipass with usb
2) enter your smartcard and pin and setup easy mode
3) setup Moolticute to use Enter + Password / Windows + L
4) got to Moolticute Database Management and add 2 Accounts to service _unlock_
5) Pull out the smartcard
6) Put it back in and enter your pin
7) after that let the Mooltipass login into your windows machine with the right account.
8) Pull out the smartcard again
9) The machine should lock itself
10) Enter smartcard again
11) Enter pin and use the same account
12) Now you getting logged in into the machine and right after that the Mooltipass getting locked
13) Smartcard still inside and the Mooltipass is locked but Windows is unlocked. 
14) Pull out the smartcard
15) Windows is getting locked again but the Mooltipass was locked already.

commit e9761fe2 (2020-06-04): fix #136