mooltipass/minible #137: Add ability to set different category for Favorites, Login and "auto-prompt"

elendil-software, 2020-06-04:
### Missing feature

Ability to set different category for Favorites, Login and "auto-prompt". For example it would be possible to select "All Categories" for the auto-prompt and other category the Favorites and/or Login menu.

### Justification

I have often situation where I want that the browser to prompt for any credential but also need to manually insert credentials (for example in a command prompt or any other software). I created a category where I put the credential I regularly need to use through Favorite/Login menu. It's easier than scrolling in a long list. But It's fastidious to change the active category every time.

### Workarounds

Change the active category.

limpkin, 2020-10-05:
out of curiosity, why don't you put the credential you need to type in the favorites menu for the "all categories" category?

elendil-software, 2020-10-05:
Good question. Probably because at this time there was this bug : https://github.com/mooltipass/minible/issues/147
Now the question is : why didn't I opened a feature request and three weeks later ? :-D

Anyway, we can close this one

limpkin, 2020-10-05:
ahaha :)

closed 2020-10-05 by limpkin