mooltipass/minible #139: Feature: Wake Up PC from Hibernate mode and use _unlock_ service

ThunderKing54, 2020-06-09:
### Missing feature
The Mooltipass should be able to wake up the computer from Hibernate mode for the _Unlock_ service.
### Justification
Tell us why you would like to see this feature added.
If I come back to my workstation most times it is in Hybernate mode. The Mooltipass is not able to wake up the workstation but it behave like every other keyboard.
### Workarounds
Press enter or any key on the normal keyboard to wake up the computer again.

limpkin, 2020-06-09:
so we would need the mini ble to detect when the computer is sleeping, then type some key to wake it up, wait a little bit, then use the standard _unlock_ feature?

ThunderKing54, 2020-06-09:
Right! :) but the Mooltipass is somehow not able to wakeup the Computer? Are the keys send over to Moolticute? That would explain why nothing happens when I send credentials in Hybernate mode and the workstation wont wake up. If the workstation is not sleeping it should send the normal keys that are configured in Moolticute.

Properrr, 2021-06-20:
@limpkin Maybe we can wait for some response from the mooltipass daemon on the host before a device could start to type credentials?

limpkin, 2021-06-20:
that would be a great idea :)