mooltipass/minible #148: Invalid Card inserted bug

ThunderKing54, 2020-06-30:
### Expected behavior
If your Mooltipass locks itself after a certain amount of time it should detect the card again after I use the scroll wheel.
### Actual behavior
If you leave your Mooltipass locked for a bit the next time you insert a card or use the scroll wheel you gonna see the message Invalid Card inserted. 
### Step by step guide to reproduce the problem
1) enter smartcard an pin
2) wait until your pc locks itself or shut it down
3) come back after 5 minutes and insert your smartcard or use the scroll wheel if the scard is still inside.
4) you gonna see the message "Invalid Card inserted"
5) If you pull out the card and put it back in the card works again

I tried it with the old card and the new one. It happens with both. I can reproduce this bug only on the newer device with the serial number on the back of it.
### Firmware Version
AUX MCU version:  0.28
Main MCU version:  0.34

### Operating System
- Windows 10 LTSB 2016

limpkin, 2020-07-01:
Hmmm I think this is something else... could you try the attached file **on your numbered unit only?**
[miniblebundle.zip](https://github.com/mooltipass/minible/files/4860453/miniblebundle.zip)

ThunderKing54, 2020-07-02:
I used your bundle file for the Mooltipass with the serial number. Unfortionally I'm not able to use my card anymore. It is always a invalid card. Even my backup card 😄

limpkin, 2020-07-02:
even when you insert it normally? :/
do you need other cards?

ThunderKing54, 2020-07-02:
I think they still gonna work but after I used your bundle I can't use any card anymore...

limpkin, 2020-07-02:
this is the numbered unit, right? can you try that one?
[miniblebundle.zip](https://github.com/mooltipass/minible/files/4862458/miniblebundle.zip)

ThunderKing54, 2020-07-02:
I try it out. Yes it is the numbered unit. BETA-0017

limpkin, 2020-07-02:
Ok! could you try that one quickly because if I made a mistake the previous firmware will drain your firmware...

ThunderKing54, 2020-07-02:
That one works good! Thank you :)

ThunderKing54, 2020-07-02:
I'm afk for about 10minutes

limpkin, 2020-07-02:
phew...
let me know if that fixes the bug as well :)

ThunderKing54, 2020-07-02:
Nope! It didn't fixed the bug.

limpkin, 2020-07-02:
but it works when normally inserting? hmm, more digging required... expect another try tonight :)

ThunderKing54, 2020-07-02:
It only happens when I lock my machine and want to use the mooltipass again or when I plug in my Mooltipass to the usb cable and insert my card for the first time. After that it only happens again if I let the Mooltipass rest for some minutes

limpkin, 2020-07-02:
Here's a long shot... could you try it?
[miniblebundle.zip](https://github.com/mooltipass/minible/files/4863458/miniblebundle.zip)

ThunderKing54, 2020-07-02:
I try it out now. Thank you :)

limpkin, 2020-07-02:
is it working? i'm crossing my fingers :D

ThunderKing54, 2020-07-02:
It is for now. But I gonna test it a bit more okay?

limpkin, 2020-07-02:
please do... it was quite a long shot :D

limpkin, 2020-07-02:
as a side note i've seen it happen on brand new devices, so i'm definitely interested in fixing that :)

ThunderKing54, 2020-07-02:
Thank you for the bundle files! It looks pretty good. Usally it happend every time after the device was locked. But now it worked 10/10 times

limpkin, 2020-07-02:
I love these long shots :)

closed 2020-07-02 by limpkin

ThunderKing54, 2020-07-02:
I know! It is awesome to fix those ;D